mani-skill/ManiSkill #1177: Missing self.tcp For UR10e

yiqiwang8177, 2025-07-10:
Hi,
When getting extra observation, [tasks ](https://github.com/haosulab/ManiSkill/blob/main/mani_skill/envs/tasks/tabletop/pull_cube.py#L117)requires tcp.  UR10e don't have it.
Why this is the case?

ms3/ManiSkill/mani_skill/envs/tasks/tabletop/push_cube.py", line 198, in _get_obs_extra
    tcp_pose=self.agent.tcp.pose.raw_pose,
             ^^^^^^^^^^^^^^
AttributeError: 'UR10e' object has no attribute 'tcp'

StoneT2000, 2025-07-10:
Yes a gripper is needed. TCP is often defined on a robot to robot basis.

yiqiwang8177, 2025-07-13:
Hi, I'm wondering is there any tutorial on combing a robotique gripper with a arm?

for ur10e, it has a .xml file. For robotique, it has .urdf. how do I combine them?

Many thanks,

yiqiwang8177, 2025-07-13:
I found a very useful link: https://github.com/culurciello/pybullet_ur5_gripper/tree/master
Problem solved