mani-skill/ManiSkill #1179: Compatibility with articulated objects in GPU parallelized environments

YingYuan0414, 2025-07-11:
I ran into `ValueError: cannot parse typestr` when trying to get qpos for the joint in an articulated object. The code works well when setting num_envs=1, but throws error with num_envs > 1 (using GPU parallelized sim). I'm using numpy 1.23.0.

Full traceback:
```
 env = gym.make(args.env_id, robot_uids=args.robot_uids, **env_kwargs)
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
  File "/home/yingyuan/dexart-release/dexhier/spraybottle.py", line 48, in __init__
    super().__init__(*args, robot_uids=robot_uids, **kwargs)
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 312, in __init__
    obs, _ = self.reset(seed=[2022 + i for i in range(self.num_envs)], options=dict(reconfigure=True))
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 890, in reset
    self._reconfigure(options)
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 728, in _reconfigure
    self._load_scene(options)
  File "/home/yingyuan/dexart-release/dexhier/spraybottle.py", line 165, in _load_scene
    self.last_openness = self.instance.get_qpos()[self.revolute_joint_index]
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/utils/structs/articulation.py", line 625, in get_qpos
    return self.qpos
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/utils/structs/articulation.py", line 754, in qpos
    return self.px.cuda_articulation_qpos.torch()[
ValueError: cannot parse `typestr`
```

As another thread of the question, there's some functions unimplemented like setting articulation properties in the GPU simulation setting. For example, 
```
File "/home/yingyuan/dexart-release/dexhier/spraybottle.py", line 171, in _load_scene
    joint.set_drive_target(0.2)  # set the target position of the joint
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/utils/structs/articulation_joint.py", line 184, in set_drive_target
    self.drive_target = target
  File "/home/yingyuan/anaconda3/envs/dexart118/lib/python3.9/site-packages/mani_skill/utils/structs/articulation_joint.py", line 240, in drive_target
    raise NotImplementedError(
NotImplementedError: Setting drive targets of individual joints is not implemented yet.
```

StoneT2000, 2025-07-11:
I can add support for setting drive positions targets for individual joints

For the first issue during the load_scene function you generally cannot access any data ahead of time, you can only load objects into the scene (and set an initial pose). qpos cannot be read or set

Environment state data like pose, qpos, qvel should be set in the _initialize_episode function, which is called at every env.reset

YingYuan0414, 2025-07-15:
Follow up on setting drive position targets: I got this error after pulling the change
```
File "/home/yingyuan/ManiSkill/mani_skill/envs/sapien_env.py", line 735, in _reconfigure
    self._load_scene(options)
  File "/home/yingyuan/dexart-release/dexhier/spraybottle.py", line 171, in _load_scene
    joint.set_drive_target(0.2)  # set the target position of the joint
  File "/home/yingyuan/ManiSkill/mani_skill/utils/structs/articulation_joint.py", line 198, in set_drive_target
    self.drive_target = target
  File "/home/yingyuan/ManiSkill/mani_skill/utils/structs/articulation_joint.py", line 254, in drive_target
    self.articulation.px.cuda_articulation_target_qpos[
TypeError: 'sapien.pysapien.CudaArray' object does not support item assignment
```

StoneT2000, 2025-07-15:
I can push a fix later but that line should be

```
self.articulation.px.cuda_articulation_target_qpos.torch()[
                self._data_index[self.scene._reset_mask[self._scene_idxs]],
                self.active_index,
            ] = arg1
```
you can edit it locally for now, ill tag this issue when the PR fixes this

YingYuan0414, 2025-07-15:
This solved my issue, thanks!

PR #1180 (2025-07-11): [Feature] Support set drive targets for individual joint
Partially closes #1179
closed 2025-07-15 by YingYuan0414