mani-skill/ManiSkill #1181: AttributeError when trying to run manual control demo

huhanwj, 2025-07-11:
Hi, I encoutered an AttributeError when trying to run `python -m mani_skill.examples.demo_manual_control -e "ReplicaCAD_SceneManipulation-v1" --render-mode="human" ` to see how to control robot in ManiSkill, but got error message:

```
opts: []
env_kwargs: {}
2025-07-11 15:50:29,201 - mani_skill  - WARNING - Mimic targets dictionary is missing for controller config for fetch. Assuming the first joint is the control joint and the second joint is the mimic joint
Observation space Box(-inf, inf, (1, 30), float32)
Action space Box(-1.0, 1.0, (12,), float32)
/home/huhan/miniconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/core.py:311: UserWarning: WARN: env.control_mode to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.control_mode` for environment variables or `env.get_wrapper_attr('control_mode')` that will search the reminding wrappers.
  logger.warn(
Control mode pd_ee_delta_pose
/home/huhan/miniconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/core.py:311: UserWarning: WARN: env.reward_mode to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.reward_mode` for environment variables or `env.get_wrapper_attr('reward_mode')` that will search the reminding wrappers.
  logger.warn(
Reward mode normalized_dense
2025-07-11 15:50:29,705 - mani_skill  - WARNING - Mimic targets dictionary is missing for controller config for fetch. Assuming the first joint is the control joint and the second joint is the mimic joint
/home/huhan/miniconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/core.py:311: UserWarning: WARN: env.agent to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.agent` for environment variables or `env.get_wrapper_attr('agent')` that will search the reminding wrappers.
  logger.warn(
Traceback (most recent call last):
  File "/home/huhan/miniconda3/envs/lerobot/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/huhan/miniconda3/envs/lerobot/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/huhan/miniconda3/envs/lerobot/lib/python3.10/site-packages/mani_skill/examples/demo_manual_control.py", line 236, in <module>
    main()
  File "/home/huhan/miniconda3/envs/lerobot/lib/python3.10/site-packages/mani_skill/examples/demo_manual_control.py", line 101, in main
    render_frame = env.render().cpu().numpy()[0]
AttributeError: 'Viewer' object has no attribute 'cpu'
``` 
The package is installed directly by pip

Is this related to GPU/cuda environment or other dependency issue? I have no clue on it. I can successfully run the random action demo in terminal with fluent running SAIPEN window.

StoneT2000, 2025-07-11:
Try removing render mode argument.

That script always opens a GUI window and expects render mode to be "sensors" or "all" or "rgb_array"

huhanwj, 2025-07-14:
> Try removing render mode argument.
> 
> That script always opens a GUI window and expects render mode to be "sensors" or "all" or "rgb_array"

That solves the issue! Thanks for the help!

closed 2025-07-14 by huhanwj