mani-skill/ManiSkill #1190: [Question] Selecting object in PickSingleYCB-v1

nicklashansen, 2025-07-16:
Hi,

I want to train single-object specialist policies for each object in the YCB dataset. I'm currently using [PickSingleYCB-v1](https://maniskill.readthedocs.io/en/v3.0.0b21/tasks/table_top_gripper/index.html#picksingleycb-v1) and find that setting `reconfiguration_freq=None` allows me to repeatedly simulate the same object. However, I haven't found a way to select which object to use using the ManiSkill API. Is this something that is possible with the code as-is or would I need to fork the repo and make those changes myself?

Any pointers would be greatly appreciated! Thanks

StoneT2000, 2025-07-16:
You would probably want to inherit the original environment and modify the logic for how assets are loaded by overriding the def __init__ function to permit passing a list of ids you want, and then to only use those ids, and then registering a new environment for each set of objects you want to have a environment ID for. This would be the clean and maintainable way.

The hacky way is easier and you don't need to make a new class and register a new env id, specifically for PickSingleYCB-v1 we sample from a list of model id string values. The list of ids available to sample from is saved here:

https://github.com/haosulab/ManiSkill/blob/3693fa7dbe827e3de0f1b87af73651f98243f767/mani_skill/envs/tasks/tabletop/pick_single_ycb.py#L65-L79

After doing gym.make just manually modify the model_ids list and then reconfigure / reset however you wish.

The json file has all the ids available as well if you want to manually look through them, they are normally saved here:
```
~/.maniskill/data/assets/mani_skill2_ycb/info_pick_v0.json
```

StoneT2000, 2025-07-16:
This probably works

```
env = gym.make("PickSingleYCB-v1")
env.unwrapped.all_model_ids = ["002_master_chef_can"]
env.reset(options=dict(reconfigure=True))
# you never need to reconfig again after that, just do normal env.reset()
```

nicklashansen, 2025-07-16:
Thank you for the quick reply! I tried the above code snippet and it appears to work :-)

closed 2025-07-16 by nicklashansen