mani-skill/ManiSkill #1209: GPU SIM parameter setting support

lyxichigoichie, 2025-07-31:
Dr. Stone,
Hi, when I simulatie a task using gpu as backend, it return a warning
```bash
[2025-07-31 12:53:49.591] [SAPIEN] [critical] PxgDynamicsMemoryConfig::collisionStackSize buffer overflow detected, please increase its size to at least 73774744 in the scene desc!
```
However, I can't retrieve this parameter at the document. So how do I set this parameter to fix this warning.

With appreciation

lyxichigoichie, 2025-07-31:
I found the [interface in sapien](https://github.com/haosulab/SAPIEN/blob/fd97712362e6f7424952d99507317d2f2606e108/include/sapien/physx/physx_default.h#L69)
```cpp
class PhysxDefault {
public:
  static std::shared_ptr<PhysxMaterial> GetDefaultMaterial();
  static void SetDefaultMaterial(float staticFriction, float dynamicFriction, float restitution);
  static void setGpuMemoryConfig(uint32_t tempBufferCapacity, uint32_t maxRigidContactCount,
                                 uint32_t maxRigidPatchCount, uint32_t heapCapacity,
                                 uint32_t foundLostPairsCapacity,
                                 uint32_t foundLostAggregatePairsCapacity,
                                 uint32_t totalAggregatePairsCapacity,
                                 uint32_t collisionStackSize);
```
But the [python interface](https://github.com/haosulab/SAPIEN/blob/fd97712362e6f7424952d99507317d2f2606e108/python/py_package/pysapien/physx.pyi#L1205) seems not to contain the `collisionStackSize` ???
```python
def set_gpu_memory_config(temp_buffer_capacity: int = 16777216, max_rigid_contact_count: int = 524288, max_rigid_patch_count: int = 81920, heap_capacity: int = 67108864, found_lost_pairs_capacity: int = 262144, found_lost_aggregate_pairs_capacity: int = 1024, total_aggregate_pairs_capacity: int = 1024) -> None:
```

StoneT2000, 2025-07-31:
Sorry I misspoke, it's not in nightly, it's in another branch.

I'll look to get it merged soon

StoneT2000, 2025-08-01:
install the latest maniskill nightly, should be fixed

closed 2025-08-01 by StoneT2000

andrewcwlee, 2025-08-01:
Thanks for your effort! 

```
[2025-08-01 16:49:25.304] [SAPIEN] [critical] PxgDynamicsMemoryConfig::collisionStackSize buffer overflow detected, please increase its size to at least 14476448 in the scene desc!
```

I think this issue still persists. I am running `UnitreeG1PlaceAppleInBowl-v1` task on gpu. 
I pulled the latest main branch.