mani-skill/ManiSkill #1210: Collision issues and unstable object states

chendl02, 2025-07-31:
I'm encountering a persistent stability issue with custom object meshes in ManiSkill 3.0.0b21. When I load custom meshes using builder.add_multiple_convex_collisions_from_file() with COACD decomposition, objects immediately start jittering on reset and sometimes get ejected at high velocity when contacted by other objects.

What I've tried:
 Adjusted COACD parameters: {"resolution": 200, "threshold": 0.1}
 Modified SceneConfig solver iterations (increase solver_position_iterations and solver_velocity_iterations to 50) and SimConfig frequencies (increase control_freq to 50)
 Used default mass/inertia parameters

Questions:
Are there specific physics parameters (contact margin, restitution, solver settings) you'd recommend for custom meshes?
Can I adjust actor stiffness in ManiSkill?
Should I use different collision shape preprocessing approaches?

For example, when the following tomato falls onto a plate in a blade-down position, it causes all the objects in the plate to bounce away

Video demonstration: https://drive.google.com/file/d/1RAB4dgyHtxUCreWYSXZOlhLp47y1qzyu/view?usp=sharing

[image]

[image]

Another video:
https://drive.google.com/file/d/1FdrOIyrWCoyb7_dPi6ITCOp9mMKI3GIo/view

StoneT2000, 2025-08-01:
Can you try increasing the physics frequency then?

Another and perhaps better option is to simplify the mesh more, it looks too detailed.

chendl02, 2025-08-01:
Simplifying the mesh significantly improves stability. 

However, when I gradually raise the physics-simulation frequency from 100 Hz to 500 Hz, the CPU backend shows an increased likelihood of objects being ejected, whereas the GPU backend remains stable.

So I'm guessing this is caused by the limited computing power of the cpu?

closed 2025-08-28 by StoneT2000