mani-skill/ManiSkill #1218: Task data generation - Cannot fetch collision meshes of merged articulations

ShiyaoExtendQA, 2025-08-04:
I was trying to generate some data for OpenCabinetDrawer-v1 task, I created a solution file and modified bash```motionplanner.py```

When I ran the motion planning, it returned the following error message:

[image]

This error message is returned from **get_actor_obb** when getting the bounding box of articulator/actor:

[image]

StoneT2000, 2025-08-05:
Yes you should access each articulation object individually to get mesh data. We currently do not provided mesh processing libraries.

You can look at the open cabinet task code to see what objects are available for access / modify it for your needs.

StoneT2000, 2025-08-07:
There is a mimic joint, one action dimension controls both gripper joints.

A robot's qpos isn't always the same dimensions as the controller actions.

If you are unsure you can do print(env.agent.controller)

to see how the currently used controller is defined and what joints it controls directly

ShiyaoExtendQA, 2025-08-07:
[image]

FredaZero, 2025-08-11:
I see. But I have one more question: why does CombinedController has 13 dof, but the when counting the dof of each group, the total number is 14 (7+1+3+3)?

StoneT2000, 2025-08-14:
Oh looks like there is a small bug. The action space is correct (13 dimensions) which is what CombinedController inspects. The sub controllers (specifically the base one) aren't reporting the correct dof. I will submit a fix for this.

PR #1226 (2025-08-14): [BugFix] fix dof value in str representation of controllers
Closes #1218
closed 2025-08-14 by StoneT2000