mani-skill/ManiSkill #1237: Motion Planning failing for SO100 arm

cmcgartoll, 2025-08-21:
Hi @StoneT2000!

Been looking to try out some SO100 training with Maniskill. Wanted to help contribute and setup the MotionPlanner + Teleop scripts for SO100 because it's helpful for me when checking out envs and simple interactions.

I was able to get it setup in my [fork](https://github.com/cmcgartoll/ManiSkill/tree/teleop-so100), and it works for some paths, but fails for a lot, especially ones involving rotation around the base motor. I tried motion planning with screw, RRTConnect, and RRTStar.

Here's an example of the working teleop:

https://github.com/user-attachments/assets/4ebc9c27-a944-4698-ac1c-7a377f175d71

Example error logs thrown by mplib for RRTConnect:
```
IK Failed! Distance 0.001396 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.002209 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.002745 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.002731 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.002099 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.001104 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.001031 is greater than threshold 0.001000.
Plan failed
IK Failed! Distance 0.001016 is greater than threshold 0.001000.
Plan failed
```

I was wondering if you've run into similar issues with mplib (or the SO100 in Maniskill) and if you know of any solutions to improve the motion planning! Thank you!

chenyenru, 2025-08-21:
Hi! Could you print and visualize the new_pose here? https://github.com/haosulab/MPlib/blob/main/mplib/planner.py#L305C21-L305C25

It would help to compare both succeeding and failing cases (determined by whether min_dist<1e9) and see for the succeeding cases whether it was the quaternion that went really off (which will verify your hypothesis).

cmcgartoll, 2025-08-27:
Thanks @StoneT2000, the solves many of the issues. Put together a PR for the teleop + MP things I worked on [here](https://github.com/haosulab/ManiSkill/pull/1245). Also thanks @chenyenru for insights

Jimntu, 2025-08-28:
> Thanks [@StoneT2000](https://github.com/StoneT2000), the solves many of the issues. Put together a PR for the teleop + MP things I worked on [here](https://github.com/haosulab/ManiSkill/pull/1245). Also thanks [@chenyenru](https://github.com/chenyenru) for insights

Fantastic! I can generate motionplanning for so100 in pick-cube task now (~50% success rate). Have you tried push cube/stack cube? If so, could you also upload the solutions for more tasks? Thanks so much!

closed 2025-08-28 by StoneT2000