mani-skill/ManiSkill #1241: Strange collision when using a mimic gripper to grasp a cube

Vector215, 2025-08-26:
When I use a DH-ag95 gripper, which is similar to the Robotiq-2f-85 gripper, to grasp a cube,Incorrect collisions occurred in the simulator.
Specifically, when the gripper contacts the block, it does not stop but continues to move inward until the gripper is closed inside the block.
I checked that the collision volume of the block and the gripper is correct. In addition, I found that when the block mass is set to light, the gripper will collide directly with the block and the block will be bounced off.
[image]
My gripper setup is as follows
```python
import sapien
import numpy as np
from mani_skill.agents.base_agent import BaseAgent, Keyframe
from mani_skill.agents.controllers import *
from mani_skill.agents.registration import register_agent
from mani_skill.sensors.camera import CameraConfig
from ..utils.get_camera_config import get_camera_param_config
from mani_skill.utils.structs.actor import Actor
from mani_skill.utils.structs.link import Link
from mani_skill.utils import common, sapien_utils
from typing import Dict, Tuple
import sapien.physx as physx
from mani_skill.agents.robots.panda import panda
import torch

# TODO: 修改urdf文件，添加一个TCP_LINK位于夹爪中间

@register_agent()
class Realman(BaseAgent):
    uid = "realman"
    urdf_path = f"/home/vector/workspace/maniskill/asset/embodiments/realman/urdf/rm65_ag95_d455_mplib.urdf"

    urdf_config= dict(
        _materials=dict(
            gripper=dict(static_friction=2.0, dynamic_friction=10.0, restitution=0.0)
        ),
        link=dict(
            gripper_finger1_finger_tip_link=dict(
                material="gripper", patch_radius=0.01, min_patch_radius=0.01
            ),
            gripper_finger2_finger_tip_link=dict(
                material="gripper", patch_radius=0.01, min_patch_radius=0.01
            ),
        ),
    )

    arm_joint_names = [
        "joint1",
        "joint2",
        "joint3",
        "joint4",
        "joint5",
        "joint6",
    ]
    gripper_joint_names = [
        "gripper_finger1_joint",  # 主控制关节
        "gripper_finger2_joint",
        "gripper_finger2_inner_knuckle_joint",
        "gripper_finger1_inner_knuckle_joint",
        "gripper_finger1_finger_tip_joint",
        "gripper_finger2_finger_tip_joint",
        "gripper_finger1_finger_joint",
        "gripper_finger2_finger_joint"
    ]
    ee_link_name = "dummy_gripper"

    arm_stiffness = 1e3
    arm_damping = 1e2
    arm_force_limit = 100

    gripper_stiffness = 1e3
    gripper_damping = 1
    gripper_force_limit = 100

[... 169 lines cut ...]
        assert np.abs(approaching @ closing) <= 1e-3
        ortho = np.cross(closing, approaching)
        T = np.eye(4)
        T[:3, :3] = np.stack([ortho, closing, approaching], axis=1)
        T[:3, 3] = center
        return sapien.Pose(T)

```
and my cube setting as below:
```python
        ab = self.scene.create_actor_builder()
        ab.set_mass_and_inertia(mass=5, cmass_local_pose=sapien.Pose([0,0,0]),inertia=[0,0,0])
        ab.add_box_collision(
            half_size=[self.cube_half_size] * 3,
        )
        ab.add_box_visual(
            half_size=[self.cube_half_size] * 3,
            material=sapien.render.RenderMaterial(
                base_color=np.array([12, 42, 160, 255]) / 255,
            ),
        )
        ab.set_initial_pose(sapien.Pose(p=[0, 0, self.cube_half_size]))
        self.cube = ab.build(name='cube')
```
I want my gripper to have correct collision after grabbing an object instead of inserting into the object. How can I do this?

Vector215, 2025-08-26:
> I don't know what happened, but maybe it's a nice try to **visualize the collision meshes** in the UI with ManiSkill, by following these steps:
> 
> 1. Make sure that you have make your environment with "render_mode='human'", so that you can get the UI
> 2. Start the simulation
> 3. Click the staff whose collision meshes you want to check out
> 4. Then find the "Articulation" bar, where you will find "Show/Hide Collision"
> 5. If it's not articulation, you should find "Entity" bar, where you will find "Static Body" or "Dynamic Body", and that's where "Show/Hide Collision" will be

In fact, I visualized the collision meshes, but it was hidden during the screenshot. I don't see any problem with the collision meshes, which confuses me.
I have read about some issues related to mimic grippers, and they have also encountered similar problems. I suspect that there may be problems with the force control and collision of the gripper itself, or problems caused by the complex structure of the gripper, But I find it difficult to pinpoint the specific issues.
I have tried adjusting the settings of stiffness, damping, and force, but it can cause problems with the gripper itself, making it difficult to find a setting that can ensure normal movement of the gripper while effectively colliding with objects.
I have also adjusted the settings for gripper collision detection without any changes.

CumulusAlpha, 2025-08-28:
hey, i think we can dig together into this question.
The mimic controller needs to be tuned when migating to nrw gripper. you can refer [ this ](https://github.com/haosulab/ManiSkill/pull/1224#issue-3307116359).
i tried this but it might not work very well for me. 
Can we work together to solve this problem, any comments will be appreciated.