mani-skill/ManiSkill #1246: Unable to replay TwoRobotPickCube-v1

ShiyaoExtendQA, 2025-08-27:
Hi, I downloaded the public TwoRobotPickCube-v1 data from huggingface. I want to get image observation, so I run the following command:
```
python -m mani_skill.trajectory.replay_trajectory --traj-path /home/er/ManiSkill/demos/TwoRobotPickCube-v1/trajectory.none.pd_joint_delta_pos.physx_cuda.h5 --save-traj --obs-mode rgb
```
However, the success rate of replaying episode is 0%

[image]

I'm wondering if this is caused by the IK difference in GPU and CPU sim? How should I replay trajectories collected in GPU sim?

StoneT2000, 2025-08-28:
IK is not used for this controller. The issue is you didn't add the option to ensure the environment start state is the same. As a result taking the same actions on different start states may lead to the wrong trajectory.

This probably fixes it
```bash
python -m mani_skill.trajectory.replay_trajectory --traj-path /home/er/ManiSkill/demos/TwoRobotPickCube-v1/trajectory.none.pd_joint_delta_pos.physx_cuda.h5 --save-traj --obs-mode rgb --use-first-env-state
```

Another note is if all you need to do is save RGB image data, you can also add `--use-env-states` instead which should give 100% replay success rate. Since you aren't generating new actions or observations different to what was recorded originally, you can skip the process of having to simulate the entire dataset to replay it.

ShiyaoExtendQA, 2025-08-30:
Thanks for your clarify. I also generated some data for OpenCabinetDrawer, and tried to replay the trajectories and save RGB image data using the following command: 
```
python -m mani_skill.trajectory.replay_trajectory --traj-path /home/er/ManiSkill/examples/baselines/ppo/runs/Mobile-Processed/OpenCabinetDrawer-v1/rl/trajectory.none.pd_joint_delta_pos.physx_cuda.h5 --save-traj --obs-mode rgb --use-env-states
```
But I got this error:

[image]

StoneT2000, 2025-09-04:
Closed in favor of [#open cab](https://github.com/haosulab/ManiSkill/issues/1259)

closed 2025-09-04 by StoneT2000