mani-skill/ManiSkill #1249: The question about mtion-planner and pd-joint-pos controller.

YZY14606, 2025-08-30:
I encountered some issues when using the motion planner and pd-joint-pos controller in ManiSkill3. I found that the end effector of my robot cannot reach the intended pose properly.

From 32 seconds to 38 seconds in the attached video, the robot's hand fails to reach the first intended pose; instead, it collides with the object midway and then moves directly toward the second pose. I am not sure whether the problem lies in the 
1. simulation environment, 
2. the motion planner,
3. the built-in controller，
4. or parameters of the motion planner and the built-in controller.

Can you give me a help? Thanks.

The images of a few key frames are as follows:

[image]

[image]

[image]

The video:

https://github.com/user-attachments/assets/70386eac-27b8-4e8e-a7a0-0c25c7d4d7a5

YZY14606, 2025-09-04:
Regarding the issue of whether any joint angle exceeds np.pi, I added the code `has_large = np.any(result["position"] > np.pi)` to the `follow_path` function of the motion planner for judgment. The results show that when similar errors occur, this judgment condition is not triggered.

StoneT2000, 2025-09-16:
I dont understand the latest video? It seems like the planner collided with the object. It probably changed to the next pose because the current plan failed/completed. A failure could be caused by the collision pointcloud being too complicated to solve for. You can print the return value of the planner, it should say something about the error.

YZY14606, 2025-09-18:
I also suspect that this might be caused by the collision. I have conducted corresponding tests, and it turns out that this situation does occur when a collision happens. However, I have already configured obstacle avoidance using a motion planner, so under normal circumstances, this should not happen.

This also is my question.

StoneT2000, 2025-09-19:
perhaps the space is too tight or the collision point cloud is not set correctly. You can try visualizing it

YZY14606, 2025-10-03:
I conducted a lot of tests, and I found that this might be due to the planner's poor performance.

I have checked the point cloud, and it was fine.

Then, I tried changing the parameter 'radius' of this function:
```
    def update_obstacle_pointcloud(self,pointcloud,radius = 0.02):
        
        self.planner.update_point_cloud(pointcloud,radius)
```
When the parameter 'radius' is set to a larger value, the number of collisions between the robotic arm and objects decreases. Conversely, when it is smaller, the frequency of collisions increases.

Now I've set the radius to 0.02, which seems to be already quite a large value (0.02m or 0.02cm?). However, my robotic arm still frequently collides with objects (about one-third of the total attempts). Are there any good solutions here?

StoneT2000, 2025-10-07:
Are you using RRT the whole time?

And could you share the pointcloud visualization? 

Small radius leading to more collisions is expected, you are essentially making more room for the robot, but can be erroneous if the points sampled are not good.

YZY14606, 2025-10-08:
When the robotic arm moves from the starting position to the first end position beneath the object, we use the RRT planner. However, a collision with the object occurred during this process.

This is pointcloud visualization:
[image]

The content in red circle is object's pointcloud. It is fine and I can't understand the reason of collision. I think that the 'radius = 0.02' is ok for motion planning.

Could this be due to the robot's movement space being too cramped? Or is it caused by the controller?