mani-skill/ManiSkill #1252: What is the correct way to set qpos of articulated objects like a drawer?

Jaraxxus-Me, 2025-09-01:
Hi, I am following the code of `OpenCabinetDrawer-v1` from `mani_skill/envs/tasks/mobile_manipulation/open_cabinet_drawer.py`.
During episode initialization, I hope to customize the open_fraction of each cabinet/drawer, may I know what would be the correct way of doing so?
I tried to take each `cabinet` as a single `Articulation` but got errors:
Thanks a lot in advance for the help!

```
        for i, model_id in enumerate(model_ids):
            # partnet-mobility is a dataset source and the ids are the ones we sampled
            # we provide tools to easily create the articulation builder like so by querying
            # the dataset source and unique ID
            cabinet_builder = get_partnet_mobility_builder(
                self.scene, f"{model_id}"
            )
            cabinet_builder.set_scene_idxs(scene_idxs=[i])
            cabinet_builder.initial_pose = sapien.Pose(p=[0, 0, 0], q=[1, 0, 0, 0])
            cabinet = cabinet_builder.build(name=f"{model_id}-{i}")
            self.remove_from_state_dict_registry(cabinet)
            # this disables self collisions by setting the group 2 bit at CABINET_COLLISION_BIT all the same
            # that bit is also used to disable collision with the ground plane
            for link in cabinet.links:
                link.set_collision_group_bit(
                    group=2, bit_idx=CABINET_COLLISION_BIT, bit=1
                )
            handle_links.append([])
            handle_links_meshes.append([])

            for link, joint in zip(cabinet.links, cabinet.joints):
                if joint.type[0] in joint_types:
                    handle_links[-1].append(link)
                    # save the first mesh in the link object that correspond with a handle
                    handle_links_meshes[-1].append(
                        link.generate_mesh(
                            filter=lambda _, render_shape: "handle"
                            in render_shape.name,
                            mesh_name="handle",
                        )[0]
                    )
            target_qlimits = handle_links[-1][0].joint.limits
            qmin, qmax = target_qlimits[..., 0], target_qlimits[..., 1]
            init_qpos = qmin + (qmax - qmin) * (self.drawer_init_open_frac[0] + self.drawer_init_open_frac[1]) / 2
            cabinet.set_qpos(init_qpos)
            self._cabinets.append(cabinet)
```

```
self = <1013-0: struct of type <class 'mani_skill.utils.structs.articulation.Articulation'>; managing 1 <class 'sapien.pysapien.physx.PhysxArticulation'> objects>, arg1 = tensor([0.4966], device='cuda:0')

    @qpos.setter
    def qpos(self, arg1: torch.Tensor):
        if self.scene.gpu_sim_enabled:
            arg1 = common.to_tensor(arg1, device=self.device)
>           self.px.cuda_articulation_qpos.torch()[
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                self._data_index[self.scene._reset_mask[self._scene_idxs]],
                : self.max_dof,
            ] = arg1
E           ValueError: cannot parse `typestr`

```

StoneT2000, 2025-09-04:
Can you try maniskill nightly or latest git commit?

Another thing is how is your code working? It should be erroring earlier since init_qpos is just a single number (target_qlimits is only the qlimits for one joint). Cabinet qpos might have a different DOF.

Lastly, you should be setting environment state data like qpos, qvel, poses in the `_initialize_episode` function for all the environments that are being reset.

StoneT2000, 2025-11-15:
Check the collision meshes?

And does this issue occur during cpu sim?

StoneT2000, 2025-11-19:
Physx has a known bug (at least the version we use) with detecting if a shape is too oblong, it detects it incorrectly and complains. Generally you can ignore it. Great to see this set of meshes worked, I'm hazarding a guess the meshes created for maniskill 1/2 weren't turns for physx gpu.