mani-skill/ManiSkill #1253: Submit a bug report

super-dashuaibi, 2025-09-02:
Hello, there's an error on line 320 of the file https://github.com/haosulab/ManiSkill/blob/main/mani_skill/utils/structs/articulation_joint.py .
I discovered this while testing how to get the joint pose. 
According to robotics, it should be self.child_link.pose * self.pose_in_child , not self.pose_in_child * self.child_link.pose , which is actually the case.

super-dashuaibi, 2025-09-10:
Okay, I'm sorry I was busy recently so I just saw

super-dashuaibi, 2025-09-10:
git push 
Username for 'https://github.com': 
Password for '
 remote: Permission to haosulab/ManiSkill.git denied to -. fatal: 无法访问 'https://github.com/haosulab/ManiSkill.git/'：The requested URL returned error: 403

Oh, it seems that I don't have the permission to submit. Maybe you need to update the code yourself.

StoneT2000, 2025-09-14:
Technically you are supposed to fork the repository, make your changes, then open a PR. I can fix it myself though today.

PR #1270 (2025-09-15): [BugFix] Fix joint pose
Closes #1253
closed 2025-09-15 by super-dashuaibi