mani-skill/ManiSkill #1267: Can't create pinocchio model on Mac

nv-quan, 2025-09-13:
Hi,

I'm trying to run ManiSkill/SAPIEN on my Mac (Apple Silicon). Everything seems to be running well except when I try to run a robot with the 'pd_ee_delta_pos' controller, it can't run because the Pinocchio model can't be created. To reproduce, simply run the `demo_robot.py` script with the option `--control-mode='pd_ee_delta_pos'`, and you'll get this error:
```bash
  File "/Users/.venv/lib/python3.11/site-packages/mani_skill/agents/controllers/utils/kinematics.py", line 88, in _setup_cpu
    ].create_pinocchio_model()
      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/.venv/lib/python3.11/site-packages/sapien/wrapper/pinocchio_model.py", line 321, in _create_pinocchio_model
    model = PinocchioModel(xml, gravity)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable
```

StoneT2000, 2025-09-14:
@fbxiang any idea?

akushonkamen, 2026-10-22:
I'd like to work on this — the docs don't mention that the IK-based controllers (pd_ee_delta_pos / pd_ee_delta_pose) require the optional pinocchio package on macOS. I'll add a note to the installation docs and a clearer error hint in the controller code path.