mani-skill/ManiSkill #1274: Curriculum learning support in ManiSkill

Yuan-Xinyi, 2025-09-20:
I would like to thank the ManiSkill team for building such a powerful and easy-to-use framework. I am currently using PPO for some tasks and was wondering whether ManiSkill supports curriculum learning. If it does, I would love to know which package should be used to implement it. If not, I am curious whether there are any plans to include curriculum learning support in future releases.

StoneT2000, 2025-09-20:
ManiSkill isn't a learning library really, we just provide a ton of examples.

For curriculum learning you will have to implement yourself at the moment.

We do happen to have one RL / online learning from demonstrations baseline called RFCL that does use a curriculum. in the baselines folder though.

Yuan-Xinyi, 2025-09-22:
Thank you so much for referring to RFCL!